gregdavill/KiBuzzard #89: Add option to specify alignment of text and placement

yaqwsx, 2022-08-16:
At the moment, it seems that the text is always aligned to the left, and the alignment option only positions the text inside the label. It would be nice to have the possibility to control both of these settings.

gregdavill, 2022-08-17:
Do you mean so the anchor can appear on the left/right of the label?
Note that the KiCad Alignment context option will align labels, without needing the origin point to be on the left/right.

[image]

yaqwsx, 2022-10-12:
Basically, what I would be interested in doing:
- create a multiline label
- specify width
- allow text-align inside the label to left, right or center. At the moment, the text is always aligned to the left, and the whole text is shifted to the left/right or center.