gregdavill/KiBuzzard #91: Cannot find KiBuzzard/buzzard/typeface directory KiCad 6

FishyG, 2022-09-14:
I apologize if this as already been asked somewhere, 

I cannot find the **"KiBuzzard/buzzard/typeface"** directory anywhere. 

I installed the KiBuzzard with KiCad's **"Plugin and Content Manager"** ([This](https://community.element14.com/resized-image/__size/1280x720/__key/communityserver-blogs-components-weblogfiles/00-00-00-00-75/6685.01_5F00_Select_5F00_Plugin_5F00_and_5F00_Content_5F00_Manager.png)) and everything seem to work, however when I tried to add custom fonts I could not find anywhere to put the fonts files.

I tried creating a new direcetory under "**/home/USER/.config/kicad/6.0/scripting/plugins" (full path /home/USER/.config/kicad/6.0/scripting/plugins/KiBuzzard/buzzard/typeface/)**

If that help, I am using **KiBuzzard 1.2.0** with **KiCad 6.0.7** on **Debian 11**
Have an awesome Day **:]**

gregdavill, 2022-09-14:
When installed via the PCM on linux the plugin should end up in `/home/<user>/.local/share/kicad/6.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/plugin.py/KiBuzzardPlugin`

FishyG, 2022-09-14:
Thank you so much :]

closed 2022-09-14 by FishyG