gregdavill/KiBuzzard #97: Feature: Add support for bars over text

FromenActual, 2022-11-30:
SparkFun's Buzzard plugin allows for bars over text to indicate inverted pins, [here's an example](https://www.sparkfun.com/products/19895) product with a couple inverted labels (MCS and ACS):

[image]

Far as I can tell, there's no way to do it with KiBuzzard yet:

[image]

Would be a great addition for feature parity!

ilium007, 2024-04-04:
Would be great to see this in the original plugin!

gregdavill, 2024-05-06:
This is now a feature thanks to #108 

[image]

closed 2024-05-06 by gregdavill