gregdavill/KiBuzzard #98: Net integration

Anathae, 2022-12-08:
Beautiful addon for KiCad.  I greatly appreciate it.

I had the silly idea that it may be able to assist with an output feature I have used in EagleCad.  Output of your plugin can be placed on the F.Cu layer, but there did not seem to be a way to associate the "footprint" with a particular net.  As such, from what I've seen, if you add a object on the F.Cu layer, and it overlaps a filled zone, the filled zone will exclude the object with the filled zone clearance values.

If KiCad allows a net association for the output of your addon, it may be possible to use it to allow text cutouts on power/ground flood fills.

I see that it is possible to add the output to either the front or the back side layers, but currently you have to edit the items properties and push it from the front to back.  Being able to select the B. layers directly from your tool may be an idea.

gregdavill, 2022-12-10:
Interesting idea... The plugin is able to find all nets in the design, so it might be something that could go onto an "advanced" tab on the plugin.

If you copy a footprint to your clipboard and paste it into a text-editor you can see the s-expr that makes up the footprint. If you can determine how to assign a net to the polygon, and how that works in s-expr that would be helpful. I suspect it would look like a dummy SMD Cu pad, then a custom shape assigned to it which is the label. But how internal/nested shapes would work I'm not sure.

I'm not sure I see the benefit of adding back-layer options to the plugin. There was a bug in an older version where editing footprints on the back layer would flip them to the front, and also adjust the rotation. But that's now been fixed. So a good workflow for lots of back-layer labels is to duplicate, then edit them.

Anathae, 2022-12-12:
Thank you for the insight to how to examine objects in KiCad.  I copied a trace segment and the text output is
```
(kicad_pcb (version 20211014) (generator pcbnew)

(layers
  (0 "F.Cu" signal)
  (31 "B.Cu" signal)
  (32 "B.Adhes" user "B.Adhesive")
  (33 "F.Adhes" user "F.Adhesive")
  (34 "B.Paste" user)
  (35 "F.Paste" user)
  (36 "B.SilkS" user "B.Silkscreen")
  (37 "F.SilkS" user "F.Silkscreen")
  (38 "B.Mask" user)
  (39 "F.Mask" user)
  (40 "Dwgs.User" user "User.Drawings")
  (41 "Cmts.User" user "User.Comments")
  (42 "Eco1.User" user "User.Eco1")
  (43 "Eco2.User" user "User.Eco2")
  (44 "Edge.Cuts" user)
  (45 "Margin" user)
  (46 "B.CrtYd" user "B.Courtyard")
  (47 "F.CrtYd" user "F.Courtyard")
  (48 "B.Fab" user)
  (49 "F.Fab" user)
  (50 "User.1" user)
  (51 "User.2" user)
  (52 "User.3" user)
  (53 "User.4" user)
  (54 "User.5" user)
  (55 "User.6" user)
  (56 "User.7" user)
  (57 "User.8" user)
  (58 "User.9" user)
)

(net 0 "")
(net 1 "GND")
(net 2 "/DP")
(net 3 "/SE")
(net 4 "/D4")
(net 5 "/D3")
(net 6 "/SD")
(net 7 "/SF")
(net 8 "/SC")
(net 9 "/SA")
(net 10 "/SG")
(net 11 "/SB")
(net 12 "/D1")
(net 13 "/D2")
(net 14 "/GP16")
(net 15 "/GP17")
(net 16 "/GP18")
(net 17 "/GP19")
(net 18 "/GP20")
(net 19 "/GP21")
(net 20 "/GP22")
(net 21 "/RUN")
(net 22 "/GP26")
(net 23 "/SW4")
(net 24 "/SW3")
(net 25 "/ADC_REF")
(net 26 "+3.3V")
(net 27 "/3V3_EN")
(net 28 "/VSYS")
(net 29 "/VBUS")
(net 30 "Net-(LED1-Pad1)")
(net 31 "Net-(LED1-Pad2)")
(net 32 "Net-(LED1-Pad6)")
(net 33 "Net-(LED1-Pad8)")
(net 34 "Net-(LED1-Pad12)")
(net 35 "/COLON")
(net 36 "/~{AM}PM")
(net 37 "/SW1")
(net 38 "/SW2")
(net 39 "Net-(LED1-Pad9)")

  (segment (start 1.11125 -1.11125) (end -1.11125 1.11125) (width 0.15875) (layer "B.Cu") (net 39) (tstamp 2f559e01-4eb0-4bbc-94f5-ab1a31ed5122))

)
```
The `(net 39)` in that nearly last line assigns it to a particular net it looks like.  

A zone, like a ground flood fill on the back layer, is a little different, It includes the net list like the trace segment but has `(net 1 (net_name "GND")` before the `(layer "B.Cu")`.  I tried adding that to the text before pasting the blizzard label back into the board editor, but KiCad flat refused to paste the resulting item.  Adding the full net list doesn't make a difference.

I'm guessing that there is a tag that is generated by your plugin that causes KiCad to reject an edited s-expr unless corrected, or perhaps removed.

As there is a way to move a label to the back layer, adding it on the plugin would strictly be for convenience.  I did have to dig a bit before I figured it out, and I'm glad I did before posting an invalid trouble item.  If it's not broke, don't fix it.

Anathae, 2022-12-12:
No idea if it will help, but the following s-expr is from a segment on the B.Cu layer associated with the GND net.
```
(kicad_pcb (version 20211014) (generator pcbnew)

(layers
  (0 "F.Cu" signal)
  (31 "B.Cu" signal)
  (32 "B.Adhes" user "B.Adhesive")
  (33 "F.Adhes" user "F.Adhesive")
  (34 "B.Paste" user)
  (35 "F.Paste" user)
  (36 "B.SilkS" user "B.Silkscreen")
  (37 "F.SilkS" user "F.Silkscreen")
  (38 "B.Mask" user)
  (39 "F.Mask" user)
  (40 "Dwgs.User" user "User.Drawings")
  (41 "Cmts.User" user "User.Comments")
  (42 "Eco1.User" user "User.Eco1")
  (43 "Eco2.User" user "User.Eco2")
  (44 "Edge.Cuts" user)
  (45 "Margin" user)
  (46 "B.CrtYd" user "B.Courtyard")
  (47 "F.CrtYd" user "F.Courtyard")
  (48 "B.Fab" user)
  (49 "F.Fab" user)
  (50 "User.1" user)
  (51 "User.2" user)
  (52 "User.3" user)
  (53 "User.4" user)
  (54 "User.5" user)
  (55 "User.6" user)
  (56 "User.7" user)
  (57 "User.8" user)
  (58 "User.9" user)
)

(net 0 "")
(net 1 "GND")
(net 2 "/DP")
(net 3 "/SE")
(net 4 "/D4")
(net 5 "/D3")
(net 6 "/SD")
(net 7 "/SF")
(net 8 "/SC")
(net 9 "/SA")
(net 10 "/SG")
(net 11 "/SB")
(net 12 "/D1")
(net 13 "/D2")
(net 14 "/GP16")
(net 15 "/GP17")
(net 16 "/GP18")
(net 17 "/GP19")
(net 18 "/GP20")
(net 19 "/GP21")
(net 20 "/GP22")
(net 21 "/RUN")
(net 22 "/GP26")
(net 23 "/SW4")
(net 24 "/SW3")
(net 25 "/ADC_REF")
(net 26 "+3.3V")
(net 27 "/3V3_EN")
(net 28 "/VSYS")
(net 29 "/VBUS")
(net 30 "Net-(LED1-Pad1)")
(net 31 "Net-(LED1-Pad2)")
(net 32 "Net-(LED1-Pad6)")
(net 33 "Net-(LED1-Pad8)")
(net 34 "Net-(LED1-Pad12)")
(net 35 "/COLON")
(net 36 "/~{AM}PM")
(net 37 "/SW1")
(net 38 "/SW2")
(net 39 "Net-(LED1-Pad9)")

  (segment (start 0 -1.27) (end 0 1.27) (width 0.25) (layer "B.Cu") (net 1) (tstamp 2192b22f-c16d-4bff-839b-dfa7d590b56e))

)
```
Forcing a refill of the ground plane for that layer doesn't cause the ground plane to add offsets for the trace, but there are no voids in the trace segment so I don't know if it will accomplish what I'm hoping.

gregdavill, 2022-12-14:
Track segments have a different format compared to footprints, The labels KiBuzzard generate are footprints.

Footprints can contain custom shaped pads, but not tracks. Maybe that's something to look further into.