gregdavill/KiBuzzard #100: Can't use on Kicad 7

Witawat, 2023-02-13:
Hello all
i try KiBuzzard but can't use on Kicad 7

ShaojieHe, 2023-02-13:
Same issue here, the plugin is unable to create new text under KiCad 7, and despite the plugin detected and trying to edit old text created under KiCad 6, but still no luck:(

When creating new text, KiCad wxPython: stdout/stderr popped out and produced the following [error](https://pastebin.com/nHXz3j5j)

I downloaded this plugin from KiCad PCM

vishnumaiea, 2023-02-14:
A workaround for the time being is to create the label with V6 and then copy-paste it to a V7 file. That worked for me.

gregdavill, 2023-02-14:
Have you tried the latest release, v1.3.0? 
https://github.com/gregdavill/KiBuzzard/releases/tag/1.3.0

It's not on the PCM yet, because I wanted to check if there were any other bugs. 
Atleast on Ubuntu this is working with v7.0.0.

ShaojieHe, 2023-02-14:
I tried downloading the 1.3.0 version of KiBuzzard and doing a manual install to scirpting/plugins folder.

this time the plugin not producing any error, but still can't create new text nor edit the existing one :(

I'm using KiCad under Windows if this info is helpful

ShaojieHe, 2023-02-14:
I also tested on Linux(Installed from flatpak) KiCad 7 but still no luck :(

Still can't create new text not edit existing one but not producing any error

ShaojieHe, 2023-02-15:
I think I found the issue, the release version of v1.3.0 not containing the compatible update for KiCad 7, the 1.3.0 [release](https://github.com/gregdavill/KiBuzzard/releases/tag/1.3.0) is 2 commits behind of master branch.

sillevl, 2023-02-15:
@ShaojieHe I just downloaded the main branch and installed it, but It did not fix anything. I am running Windows.

sillevl, 2023-02-15:
> @ShaojieHe I just downloaded the main branch and installed it, but It did not fix anything. I am running Windows.

Seemed I needed to install it in `/3rdparty/plugins` instead of `/scripting/plugins`. Now it is working again.

tronixio, 2023-02-19:
Download main ([4dfddeb](https://github.com/gregdavill/KiBuzzard/commit/4dfddeb3cb32278c0b147cbc6d2fc621a3697f05)) in `~/Documents/KiCad/7.0/3rdparty/plugins`.
Will fix this issue, OS X Ventura 13.2.1 (intel), Kicad 7.0.0-0

1.3.0 installed manually or with PCM doesn't work

ccdkid, 2023-02-21:
> I think I found the issue, the release version of v1.3.0 not containing the compatible update for KiCad 7, the 1.3.0 [release](https://github.com/gregdavill/KiBuzzard/releases/tag/1.3.0) is 2 commits behind of master branch.
> 
> Just download the plugin from the main branch and do a manual install will fix this issue

I installed it via pcm and afterwards i just copied all files from the master branch into `\Documents\KiCad\7.0\3rdparty\plugins\com_github_gregdavill_KiBuzzard`

gregdavill, 2023-02-21:
I've made a new release: https://github.com/gregdavill/KiBuzzard/releases/tag/1.4.0

I'll get this merged into the PCM soon.

technoo10201, 2023-02-23:
> Download main ([4dfddeb](https://github.com/gregdavill/KiBuzzard/commit/4dfddeb3cb32278c0b147cbc6d2fc621a3697f05)) in `~/Documents/KiCad/7.0/3rdparty/plugins`. Will fix this issue, OS X Ventura 13.2.1 (intel), Kicad 7.0.0-0
> 
> 1.3.0 installed manually or with PCM doesn't work

Hi guys, same here with Ubuntu. I installed the plugin from the PCM, then i replaced the files with the main branch. Still not working

technoo10201, 2023-02-23:
> > I think I found the issue, the release version of v1.3.0 not containing the compatible update for KiCad 7, the 1.3.0 [release](https://github.com/gregdavill/KiBuzzard/releases/tag/1.3.0) is 2 commits behind of master branch.
> > Just download the plugin from the main branch and do a manual install will fix this issue
> 
> I installed it via pcm and afterwards i just copied all files from the master branch into `\Documents\KiCad\7.0\3rdparty\plugins\com_github_gregdavill_KiBuzzard`
> 
> now, works like a charm ;)
> 
> Windows 11

Hi, I installed (Ubuntu) the plugin (Kicad 7) from the PCM, then i replaced the files with the main branch. Still not working

voltlog, 2023-02-26:
Running on WIN10 Kicad 7.0, tried both kibuzzard 1.3 and 1.4, labels I create are invisible, nothing shows up after clicking OK in the dialog box.
[image]

gregdavill, 2023-02-27:
I'm having a hardtime replicating this issue. I've tried the 1.4.0 release on both Windows and Ubuntu KiCad 7.0 without issue. :/
If anyone is having issues could the attach the kibuzzard.log file? This is currently created in the kibuzzard plugin folder, and might flag the potential issue.

@voltlog after clicking okay, does "Ctrl + V" result in the label appearing? Just trying to determine if it's being created but not placed, or not created at all.

voltlog, 2023-02-27:
@gregdavill "Ctrl +V" does not result in label appearing, forgot to mention but I already tried that. I have attached my log file.
[kibuzzard.log](https://github.com/gregdavill/KiBuzzard/files/10839529/kibuzzard.log)

dzalf, 2023-02-27:
Same issue here, guys. On Windows, KiCAD 7.0 the label does not show up. Trying what others have reported...

dzalf, 2023-02-27:
> Same issue here, guys. On Windows, KiCAD 7.0 the label does not show up. Trying what others have reported...

Ok, I found a solution (for Windows 11).

With KiCAD closed:

1. Download the latest release.
2. Unzip the file inside the downloaded location.
3. Take the folder "KiBuzzard" from inside the unzipped folder
4. Paste it inside: inside C:\Users\~USER~\Documents\KiCad\7.0\scripting\plugins
5. Run KiCAD PCB Editor.
6. Now you can create and edit labels no problem

If you had it preinstalled via PCM, uninstall it (I couldn't make it work without removing it from the manager)

Cheers 🍻

voltlog, 2023-03-04:
> 

I confirm, method described by @dzalf above works on my setup as well (WIN 10 PRO with Kicad 7.0)

dh1tw, 2023-03-12:
I can confirm that >1.4.0 fixes the mentioned problem on MacOS Ventura (M1 CPU).

Make sure you don't have KiBuzzard installed through the plugin manager. If so, uninstall it.

Then clone the repo into the plugins directory:
```
$ cd /Users/tobias/Documents/KiCad/7.0/scripting/plugins
$ git clone https://github.com/gregdavill/KiBuzzard.git
```

Reload the plugins and you are ready to go.

0x3333, 2023-03-23:
> Can also confirm that >1.4.0 fixes, M1 Mac Book Pro, OSX Ventura
Same here, but on Monterey

itavero, 2023-03-26:
For me using the master branch also mostly worked on Mac OS X Monterey (Intel Mac) with KiCad 7.0.0.

One thing that doesn't work for me is creating labels that have a rounded Cap (e.g. `(` or `)`). These fail to generate (pasting also doesn't help them appear).

kawashimaken, 2023-04-19:
I had the same problem, 
My env is KiCad 7.0.2 /Windows 11 pro

My solution is the  following, hope it helps someone:
`$ cd  C:/Users/myusername/Documents/KiCad/7.0/3rdparty/plugins
$ git clone https://github.com/gregdavill/KiBuzzard.git`

restart the KiCad, and it works now.

itavero, 2023-04-19:
> I had the same problem, 
> 
> My env is KiCad 7.0.2 /Windows 11 pro

Your solution is the same as is mentioned earlier in this thread. Can you also create labels with a rounded Cap (e.g. `(` or `)`)?

nickoe, 2023-04-19:
The 1.4.0 version is tagged, when will a new update be submitted to the kicad addons? @gregdavill  Is there some gotcha with the tag that I missed?

kawashimaken, 2023-04-20:
> > I had the same problem,
> > My env is KiCad 7.0.2 /Windows 11 pro
> 
> Your solution is the same as is mentioned earlier in this thread. Can you also create labels with a rounded Cap (e.g. `(` or `)`)?

Yes, I can create labels with a rounded Cap.

frippe75, 2023-04-23:
On Fedora37 using Kicad 7.0.0-rc1-unknown-0.5.rc2.fc36

Under the 3rdparty folder there wasn't a plugins folder so I created one and performed a git clone. But plugin is not being picked up.

haakonnessjoen, 2023-05-05:
On Mac OSX, Ventura 13.1. Using KiCad 7.0.2-0. I only saw KiBuzzard 1.3.0 in the main repo, and that didn't work.
I uninstalled it, and cloned the master branch to my ~/Documents/KiCad/7.0/scripting/plugins folder, and that worked.

pcfreak1201, 2023-08-14:
Hi all!
On Linux (Mint 21.2 Victoria / base: Ubuntu 22.04 jammy) it doesn't work on a new machine with all updates and KiCad 7 (Version: 7.0.7-7.0.7~ubuntu22.04.1, release build)
The log says:
```
08-14 23:15:04 svg2mod 1301:Font "mplus-1mn-medium" does not natively support style "Regular" using "Path" instead
08-14 23:15:04 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:04 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:04 svg2mod 304:  Inlining 1 segments...
08-14 23:15:04 svg2mod 339:  Writing Text with 25 points
08-14 23:15:04 svg2mod 1301:Font "mplus-1mn-medium" does not natively support style "Regular" using "Path" instead
08-14 23:15:04 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:04 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:04 svg2mod 304:  Inlining 4 segments...
08-14 23:15:04 svg2mod 339:  Writing Text with 59 points
08-14 23:15:05 svg2mod 1301:Font "mplus-1mn-medium" does not natively support style "Regular" using "Path" instead
08-14 23:15:05 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:05 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:05 svg2mod 304:  Inlining 4 segments...
08-14 23:15:05 svg2mod 339:  Writing Text with 59 points
08-14 23:15:05 svg2mod 1301:Font "mplus-1mn-medium" does not natively support style "Regular" using "Path" instead
08-14 23:15:05 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:05 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:05 svg2mod 304:  Inlining 5 segments...
08-14 23:15:05 svg2mod 339:  Writing Text with 71 points
08-14 23:15:06 svg2mod 1301:Font "mplus-1mn-medium" does not natively support style "Regular" using "Path" instead
08-14 23:15:06 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:06 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:06 svg2mod 304:  Inlining 8 segments...
08-14 23:15:06 svg2mod 339:  Writing Text with 167 points
08-14 23:15:06 svg2mod 339:  Writing Text with 9 points
08-14 23:15:11 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
08-14 23:15:11 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:11 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:11 svg2mod 304:  Inlining 8 segments...
08-14 23:15:11 svg2mod 339:  Writing Text with 202 points
08-14 23:15:11 svg2mod 339:  Writing Text with 10 points
08-14 23:15:15 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
08-14 23:15:15 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:15 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:15 svg2mod 304:  Inlining 8 segments...
08-14 23:15:15 svg2mod 339:  Writing Text with 202 points
08-14 23:15:15 svg2mod 339:  Writing Text with 10 points
08-14 23:15:20 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
08-14 23:15:20 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:20 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:20 svg2mod 304:  Inlining 8 segments...
08-14 23:15:20 svg2mod 339:  Writing Text with 233 points
08-14 23:15:20 svg2mod 339:  Writing Text with 10 points
08-14 23:15:23 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
08-14 23:15:23 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:23 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:23 svg2mod 304:  Inlining 8 segments...
08-14 23:15:23 svg2mod 339:  Writing Text with 234 points
08-14 23:15:23 svg2mod 339:  Writing Text with 10 points
08-14 23:15:34 com_github_gregdavill_KiBuzzard.plugin 57:Running KiBuzzard
08-14 23:15:34 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
08-14 23:15:34 svg2mod 263:  Detailed names: [F.SilkS]
08-14 23:15:34 svg2mod 304:  Inlining 8 segments...
08-14 23:15:34 svg2mod 339:  Writing Text with 234 points
08-14 23:15:34 svg2mod 339:  Writing Text with 10 points
08-14 23:15:34 com_github_gregdavill_KiBuzzard.plugin 71:Loading label onto clipboard
08-14 23:15:34 com_github_gregdavill_KiBuzzard.plugin 146:No pcbnew window found
```
Anything I can do / test?

gregdavill, 2023-08-15:
@pcfreak1201 Can you paste a footprint into the board layout? It looks like it's put a footprint on the clipboard.
Which version of KiBuzzard are you using?

pcfreak1201, 2023-08-16:
Huh - YES! It works! I can paste the footprint by myself. May be a language setting problem? I use german descriptions and there is no pcbnew on the window.
I use(d) the latest by the manager, but also overwrite the version from github (master). No changes.

prasimix, 2024-01-06:
I tried both v1.4 and v1.5 but it seems that it doesn't work on Ubuntu Release 22.04.3 LTS (Jammy Jellyfish) 64-bit, Kernel Linux 6.2.0-39-generic x86_64

pcfreak1201, 2024-01-06:
Hmm, on 
KiCad 7.0.10-7.0.10~ubuntu22.04.1, release build on my Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy, Kernel 5.15
it still works. So because I'd lots of problems with kernel 6 on my other systems, could you switch back your kernel at boot?

gregdavill, 2024-05-06:
Closing as KiCad 7 support has worked for awhile. 
If you encounter issues it's likely outside of KiCad7, please open a new issue.

closed 2024-05-06 by gregdavill