gregdavill/KiBuzzard #104: DRC warning: "The current configuration does not include the library ''."

xobs, 2023-05-20:
With the latest `main` version da33521870e1b51d625efb8dc190ef643dab437f, each instance of kibuzzard generates a DRC failure:

[image]

This is with a freshly-installed KiCad 7.0.2

joseluislemus, 2023-09-04:
Same issue here. This is with KiCad 7.0.7 on MacOS Ventura.

gregdavill, 2023-09-04:
This issue is mainly because the labels are generated, rather than coming from a library. IMO this doesn't matter especially since the labels are saved in KiCad project. But I understand the rationale behind the DRC check for this.

There are a few ways to fix this:
 - Manually mark each KiBuzzard label as excluded
 - Disabling that DRC check, or switch it to a warning
 - Extend KiBuzzard so that it saves each label into a global/project library.

I'm not sure it really makes sense to do the last item.

closed 2024-05-06 by gregdavill