gregdavill/KiBuzzard #110: Error with KiCad 8

boromyr, 2024-01-29:
```py
Errore KiCad

[Content]
Traceback (most recent call last):
  File "D:\Users/PC/Documents/KiCad/8.0/scripting\plugins\com_github_gregdavill_KiBuzzard\plugin.py", 
	line 93, in run_buzzard
    io = pcbnew.PCB_PLUGIN()
         ^^^^^^^^^^^^^^^^^
AttributeError: module 'pcbnew' has no attribute 'PCB_PLUGIN'

```
When I try to modify an existing label, I get this error. I tried to search for `PCB_PLUGIN` in pcbnew.py of version 8 of KiCad and I believe it has been replaced with `PCB_IO`.

Windows 11 
KiCad 8 RC2
KiBuzzard https://github.com/gregdavill/KiBuzzard/commit/207b9701f8974d12ba3a5e0ece8fa42601168edd

arturo182, 2024-01-30:
Can confirm in `8.0.0-rc2-253-g7e194d5f69` on Windows 11

[image]

Only affects editing, can still add new labels.

boromyr, 2024-01-30:
Solved, just replace `io = pcbnew.PCB_PLUGIN()` with `io = pcbnew.PCB_IO_KICAD_SEXPR()`

PR #111 (2024-02-08): Fix editing with Kicad v8 RC
Fixes #110
commit 825943be (2024-02-08): Fix editing with Kicad v8 RC

Fixes #110
closed 2024-02-08 by arturo182