gregdavill/KiBuzzard #112: Kicad 8 ?

PricelessToolkit, 2024-02-24:
It seems not working with Kicad 8, I can't place any text, can someone test it? Thanks

arturo182, 2024-02-24:
@gregdavill I think it's time for a new release now that v8 is stable. But first #99 needs to be fixed or reverted. I sense a lot more people coming here in the following days complaining about the plugin not working in v8 😅

ha11otronix, 2024-02-25:
> @gregdavill I think it's time for a new release now that v8 is stable. But first #99 needs to be fixed or reverted. I sense a lot more people coming here in the following days complaining about the plugin not working in v8 😅

Agree ! As I just installed and first I wanted to try was this tool via PCM and came to know that it's not working so for now I have reverted my KiCad version.

arturo182, 2024-02-25:
While Greg finds time and makes the decision, I pushed a [v8 branch](https://github.com/gregdavill/KiBuzzard/tree/v8) with the breaking commit reverted so anyone in need can clone or download it and manually replace the plugin and have it work in v8.

craigarno, 2024-02-27:
@arturo182, I tried your branch under Windows 11 and was unable to get it to function.  There is a good possibility I did something wrong, but I did try to follow your directions.  This was the result:
[image]

I tried creating a KiBuzzard label, nothing appeared.  So I tried the usual [ctrl]+V, and nothing.  Tried "Paste", nothing.

KiBuzzard Labels from my saved KiCad v7 project edited using KiCad v8 can be selected, modified, but not saved to the .kicad_pcb file.

Since other help requests are prompted to include the kibuzzard.log file, I'm attaching mine.  
[kibuzzard.log](https://github.com/gregdavill/KiBuzzard/files/14412785/kibuzzard.log)

This file may be a mashup of KiCad v7 KiBuzzard logging and any logging done by your KiBuzzard modifications.  Hope it helps.

This reporting is just to say KiBuzzard and workarounds aren't working with my Windows 11/KiCad v8rc3 installation.

No rush, I'll keep using KiCad v7 for now, which I'd planned to do anyway until v8 issues settle down.

```
Application: KiCad x64 on x64

Version: 8.0.0-rc3, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.3.0
	FontConfig 2.14.2
	libcurl/8.5.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 16 2024 16:09:39
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.5.0-DEV
	ngspice: 42
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
```

gregdavill, 2024-02-27:
@craigarno Your last lines in the log file have the following error:
```
02-26 16:26:47 com_github_gregdavill_KiBuzzard.plugin 107:Version check failed "8.0.0-rc3" not in version list
```
I think when you unzipped the KiBuzzard-8 zip it put all the files into a folder `kibuzzard-8` instead of overwriting the old files with updated versions.

PricelessToolkit, 2024-02-27:
Thank you very much, 8 works.

arturo182, 2024-02-28:
@gregdavill I think we're ready for a release 🫡

marshallbrown123, 2024-03-04:
I installed this on the release version of Kicad 8 through the package manager -I have the same issue.

03-04 22:22:38 svg2mod 1301:Font "BerkeleyMono-Bold" does not natively support style "Regular" using "Path" instead
03-04 22:22:38 svg2mod 1301:Font "BerkeleyMono-Bold" does not natively support style "Regular" using "Path" instead
03-04 22:22:38 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
03-04 22:22:38 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
03-04 22:22:38 svg2mod 263:  Detailed names: [F.SilkS]
03-04 22:22:38 svg2mod 263:  Detailed names: [F.SilkS]
03-04 22:22:38 svg2mod 304:  Inlining 6 segments...
03-04 22:22:38 svg2mod 304:  Inlining 6 segments...
03-04 22:22:38 svg2mod 339:  Writing Text with 90 points
03-04 22:22:38 svg2mod 339:  Writing Text with 90 points
03-04 22:22:39 com_github_gregdavill_KiBuzzard.plugin 57:Running KiBuzzard
03-04 22:22:39 com_github_gregdavill_KiBuzzard.plugin 57:Running KiBuzzard
03-04 22:22:39 com_github_gregdavill_KiBuzzard.plugin 107:Version check failed "8.0.0" not in version list
03-04 22:22:39 com_github_gregdavill_KiBuzzard.plugin 107:Version check failed "8.0.0" not in version list
03-04 22:22:39 com_github_gregdavill_KiBuzzard.plugin 148:Version check failed "8.0.0" not in version list
03-04 22:22:39 com_github_gregdavill_KiBuzzard.plugin 148:Version check failed "8.0.0" not in version list

I'm unsure how I'm supposed to work around it? are there any instructions on how to fix it. 
Thanks.

arturo182, 2024-03-04:
Read the comments above

marshallbrown123, 2024-03-05:
Sorry - I assumed that the package manager had an updated version that worked. 

I've followed the howto above and it's working now,

CrackXT, 2024-03-07:
Hello everyone, I have "installed" the pre-release version 1.6.0 according to arturo182's description. Nevertheless, there seems to be a problem. The last line in the log says there is no window open :( but the cache contains the generated label. So that you can at least display the generated label by inserting it.

...
03-07 13:00:49 com_github_gregdavill_KiBuzzard.plugin 71:Loading label onto clipboard
03-07 13:00:49 com_github_gregdavill_KiBuzzard.plugin 150:No pcbnew window found

nullstalgia, 2024-03-22:
Seems to be cropping up with KiCad 8.0.1 after working fine on 8.0.0.

`03-22 02:09:52 com_github_gregdavill_KiBuzzard.plugin 148:Version check failed "8.0.1" not in version list`

EmielV2002, 2024-04-28:
I'm not sure if this is still relevant but downloading the [https://github.com/gregdavill/KiBuzzard/releases/tag/1.6.0](
KiBuzzard-1.6.0-pcm.zip). Removing the old KiBuzzard installation and reinstalling KiBuzzard with the downloaded file above via the KiCad package manager works. 

I'm using KiCad 8.0.1 on Windows 11.

craigarno, 2024-05-05:
> I'm not sure if this is still relevant but downloading the [KiBuzzard-1.6.0-pcm.zip](https://github.com/gregdavill/KiBuzzard/releases/tag/1.6.0). Removing the old KiBuzzard installation and reinstalling KiBuzzard with the downloaded file above via the KiCad package manager works.

I tried this with my design using KiCad 8.0.2 and Windows 11.  A quick check (not exhaustive) suggests this KiBuzzard installation procedure worked for me.

MichaDit, 2024-06-08:
Here not working, get the following log.
At least manual insert works (STRG+V) after generating Label.
Win10, KiCad Version: 8.0.2, release build, KiBuzzard 1.6.0 via Plugin-Manager
```
06-08 16:37:24 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
06-08 16:37:25 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
06-08 16:37:25 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
06-08 16:37:25 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
06-08 16:37:25 svg2mod 263:  Detailed names: [F.SilkS]
06-08 16:37:25 svg2mod 351:  Writing Text with 9 points
06-08 16:37:25 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
06-08 16:37:25 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
06-08 16:37:25 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
06-08 16:37:25 svg2mod 263:  Detailed names: [F.SilkS]
06-08 16:37:25 svg2mod 339:  Writing Text with 9 points
06-08 16:37:25 svg2mod 339:  Writing Text with 28 points
06-08 16:37:25 svg2mod 304:  Inlining 1 segments...
06-08 16:37:25 svg2mod 339:  Writing Text with 47 points
06-08 16:37:25 svg2mod 339:  Writing Text with 43 points
06-08 16:37:27 com_github_gregdavill_KiBuzzard.plugin 57:Running KiBuzzard
06-08 16:37:27 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
06-08 16:37:27 svg2mod 263:  Detailed names: [F.SilkS]
06-08 16:37:27 svg2mod 339:  Writing Text with 9 points
06-08 16:37:27 svg2mod 339:  Writing Text with 28 points
06-08 16:37:27 svg2mod 304:  Inlining 1 segments...
06-08 16:37:27 svg2mod 339:  Writing Text with 47 points
06-08 16:37:27 svg2mod 339:  Writing Text with 43 points
06-08 16:37:27 com_github_gregdavill_KiBuzzard.plugin 71:Loading label onto clipboard
06-08 16:37:27 com_github_gregdavill_KiBuzzard.plugin 150:No pcbnew window found
```

gregdavill, 2024-06-09:
@MichaDit this might be fixed by #123