gregdavill/KiBuzzard #121: Improve Height calculation

Drachenkaetzchen, 2024-05-07:
It seems the height is taken from the bounding box, which ends up in odd heights. Here's an example:

These are the settings I used:

[image]

Then I put 2 labels next to each other, one with the text `max 2A` and one with the text `max 2Aj`. As you can see, both the `max 2Aj` label's width and height are smaller.

[image]

gregdavill, 2024-05-07:
Yes this is a known issue, #115 We've not rolled the fix into a release just yet, but a fix is available.
Can you try the latest commit? 6d79f60b 

Now the text `2A` will be scaled to 1.3mm in height. I'm not sure we have a good solution to handle characters below the baseline yet.
[image]

Drachenkaetzchen, 2024-05-07:
Thank you, I missed that issue!

I'll try it out tomorrow or in the next few days - can we keep this issue open so I don't forget to check?

Regarding the baseline issue, I understand that's tricky. How about setting an overall height of the label? I think there should be two items: "define height" and "set font size". If I set the height to, let's say, 1.3mm, and choose a font size which will exceed the 1.3mm height when I use "j", I know that I should probably reduce the font size.