gregdavill/KiBuzzard #125: Variables don't work

lukilukeskywalker, 2025-01-18:
When trying to set a variable, it doesn't generate the KiBuzzard Label.

[image]

Would be cool to be able to use Variables for stuff like Revision numbers and other stuff

lukilukeskywalker, 2025-01-18:
Hmm, I noticed that it does in fact generate the KiBuzzard Label, but it doesn't change to the value of the variable

[image]
(I would expect something like the upper text, but just get the text I wrote)

gregdavill, 2025-02-07:
This might be possible.But would rely on:
 - Python having access to the variable substitution. (I'm unclear if this possible)

It would have the following limitation:
 - Wouldn't update the text when the referenced variable is changed. This is a limitation on the plugin system.

eg. `${LAYER}` could potentially be set to `F.Silk`. But when the label is flipped to B.Silk, it would still retain the text `F.Silk`
If you needed the label to update, you might need to use the built-in KiCad label facilities.

lukilukeskywalker, 2025-02-10:
I think the main use for such a function would be to set stuff like board revision and Date, or something like that.
Maybe a update button can be added to update all Kibuzzards to display the latest variation? 
No idea if that is possible
It would be cool if everytime B is pressed to rebuild fill zones the Kibuzzards would update, but no idea if that is even feasible