gregdavill/KiBuzzard #129: Text not placed in KiCad 8.08, KiBuzzard 1.6

ThePurox, 2025-02-08:
## Bug description
When I'm trying to place text with KiBuzzard onto the PCB, the text gets rendered in the preview window, but not placed on the PCB.
The command line logs point to similar issues as #112 , but furthermore KiBuzzard does not find the pcbnew window.

## Steps to reproduce
Open empty KiCad project.
Open PCB Editor
Click on "Add Label"-Button by KiBuzzard
enter text (text appears in preview)
click on OK
(no text appears in PCB Editor window)

## Version info: 
KiBuzzard: 1.6.0 installed via KiCad Plugin manager
Python: 3.13.1 
KiCad: Version: 8.0.8, release build

KiCad x86_64 on x86_64

Version: 8.0.8, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.2.0
	FontConfig 2.15.0
	libcurl/8.11.1 OpenSSL/3.4.0 zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 libidn2/2.3.7 libpsl/0.21.5 libssh2/1.11.1 nghttp2/1.64.0 nghttp3/1.7.0

Platform: Arch Linux, 64 bit, Little endian, wxGTK, X11, awesome, x11

Build Info:
	Date: Jan 17 2025 23:21:53
	wxWidgets: 3.2.6 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.11.1
	ngspice: 43
	Compiler: GCC 14.2.1 with C++ ABI 1019

Build settings:
	KICAD_USE_EGL=ON

## Logs:
```
/home/nico/.local/share/kicad/8.0/3rdparty/plugins/com_github_gregdavill_KiBuzzard/dialog/compat.py:17: wxPyDeprecationWarning: Call to deprecated item. Use SetSizeHints instead.
  super(wx.Dialog, self).SetSizeHintsSz( a,b)
Font "FreddySpark-Regular" does not natively support style "Regular" using "Path" instead
02-08 09:52:00 svg2mod 1301:Font "FreddySpark-Regular" does not natively support style "Regular" using "Path" instead
Font "FreddySpark-Regular" does not natively support style "Regular" using "Path" instead
02-08 09:52:12 svg2mod 1301:Font "FreddySpark-Regular" does not natively support style "Regular" using "Path" instead
Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
02-08 09:52:17 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
02-08 09:52:31 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
02-08 09:52:38 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
02-08 09:52:38 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
Found SVG layer: F.SilkS
02-08 09:52:38 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
  Detailed names: [F.SilkS]
02-08 09:52:38 svg2mod 263:  Detailed names: [F.SilkS]
  Inlining 2 segments...
02-08 09:52:38 svg2mod 304:  Inlining 2 segments...
  Writing Text with 78 points
02-08 09:52:38 svg2mod 339:  Writing Text with 78 points
  Writing Text with 10 points
02-08 09:52:38 svg2mod 339:  Writing Text with 10 points
Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
02-08 09:52:38 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
02-08 09:52:38 svg2mod 1301:Font "UbuntuMono-B" does not natively support style "Regular" using "Path" instead
Found SVG layer: F.SilkS
02-08 09:52:38 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
  Detailed names: [F.SilkS]
02-08 09:52:38 svg2mod 263:  Detailed names: [F.SilkS]
  Inlining 4 segments...
02-08 09:52:38 svg2mod 304:  Inlining 4 segments...
  Writing Text with 154 points
02-08 09:52:38 svg2mod 339:  Writing Text with 154 points
  Writing Text with 10 points
02-08 09:52:38 svg2mod 339:  Writing Text with 10 points
02-08 09:52:50 com_github_gregdavill_KiBuzzard.plugin 57:Running KiBuzzard
Found SVG layer: F.SilkS
02-08 09:52:50 svg2mod-unfiltered 262:Found SVG layer: F.SilkS
  Detailed names: [F.SilkS]
02-08 09:52:50 svg2mod 263:  Detailed names: [F.SilkS]
  Inlining 4 segments...
02-08 09:52:50 svg2mod 304:  Inlining 4 segments...
  Writing Text with 154 points
02-08 09:52:50 svg2mod 339:  Writing Text with 154 points
  Writing Text with 10 points
02-08 09:52:50 svg2mod 339:  Writing Text with 10 points
02-08 09:52:50 com_github_gregdavill_KiBuzzard.plugin 71:Loading label onto clipboard
02-08 09:52:50 com_github_gregdavill_KiBuzzard.plugin 150:No pcbnew window found
```

gregdavill, 2025-02-11:
When your PCB editor is open, what is the window title?
We have had issues in the past trying to match the text "PCB Editor" directly on non-english systems.

gregdavill, 2025-02-11:
We currently pass-through `wx.GetTranslation()` which was a fix for the locale issue. But I wonder if that's causing an issue here.

If you can open up a scripting console from the PCB editor (Tools->"Scripting Editor"), you can check what's returned by copying each line, and running them 1 at a time.
```python
>>> import wx
>>> [x for x in wx.GetTopLevelWindows() if (wx.GetTranslation('pcbnew') in x.GetTitle() and not 'python' in x.GetTitle().lower()) or (wx.GetTranslation('PCB Editor') in x.GetTitle())]
>>> [x.GetTitle() for x in wx.GetTopLevelWindows()]
>>> wx.GetTranslation('PCB Editor')
```

After importing wx, the first command is run to find the reference to the PCB Editor window, it should print out an array of 1 item if it finds the window.
Since that's likely not happening, the last 2 commands will help diagnose a fix.

ThePurox, 2025-02-11:
I get weird errors, that the `'WindowList_iterator' object is not iterable`... (I tried a reboot of my laptop, but the errors remain)
However I can access them with their indices directly.
See the commands and their output below.

I can follow your logic in the first command. Given the output of the commands, (once I do the manual indexing) it should work. I however have no idea, why the `WindowList_iterator` is not iterable.

```
Py 0.9.8
Python 3.13.1 (main, Dec  4 2024, 18:05:56) [GCC 14.2.1 20240910] on linux
Type "help", "copyright", "credits" or "license" for more information.
Startup script executed: /home/nico/.config/kicad/8.0/PyShell_pcbnew_startup.py
>>> import wx
[x for x in wx.GetTopLevelWindows() if (wx.GetTranslation('pcbnew') in x.GetTitle() and not 'python' in x.GetTitle().lower()) or (wx.GetTranslation('PCB Editor') in x.GetTitle())]
Traceback (most recent call last):
  File "<input>", line 1, in <module>
TypeError: 'WindowList_iterator' object is not iterable
>>> [x.GetTitle() for x in wx.GetTopLevelWindows()]
Traceback (most recent call last):
  File "<input>", line 1, in <module>
TypeError: 'WindowList_iterator' object is not iterable
wx.GetTranslation('PCB Editor')
'PCB Editor'
>>> wx.GetTopLevelWindows()
WindowList: [<wx._core.Frame object at 0x75b63d4f2570>, <wx._core.Frame object at 0x75b63d4f27b0>, <wx._core.Frame object at 0x75b63d4f28d0>, <wx._core.Frame object at 0x75b63d44c4d0>]
>>> wx.GetTopLevelWindows()[1]
<wx._core.Frame object at 0x75b63d4f23c0>
>>> tlws = wx.GetTopLevelWindows()
>>> for i in range(len(tlws)):
>>>     tlws[i]
>>>     tlws[i].GetTitle()
    
<wx._core.Frame object at 0x75b63d4f28d0>
'space-invader — KiCad 8.0'
<wx._core.Frame object at 0x75b63d4f28d0>
'space-invader — PCB Editor'
<wx._core.Frame object at 0x75b63d4f28d0>
''
<wx._core.Frame object at 0x75b63d44c4d0>
'KiPython'
>>> wx.GetTranslation('PCB Editor')
'PCB Editor'
```

gregdavill, 2025-02-11:
Strange indeed, I've adjusted the logic, and opened a PR with the changes if you're able to try them out.

riacob, 2025-02-22:
The same issue is present in KiCad 9. Will you accept @ThePurox pull request #130? Or are you working on another solution?

gregdavill, 2025-02-22:
We can merge that in, I think it's an issue with particular wxWidgets versions. The solution is backwards compatible. We'll probably have to push a release soon so that KiCad 9 users can run this from the PCM.

gregdavill, 2025-03-01:
1.6.1 was released. A package to install via the PCM is in the release: https://github.com/gregdavill/KiBuzzard/releases/tag/1.6.1

It should be available directly in the PCM in the coming days. (Merged upstream: https://gitlab.com/kicad/addons/metadata/-/commit/205562d9d347971cfd21cb41aa943b5b5f165866 )