gregdavill/KiBuzzard #131: Labels not changing on project ported from K8.08 to K9.00

dzalf, 2025-02-25:
Dear @gregdavill 

Thank you for the f**ing fantastic job you have done with KiBuzzard

Unfortunately, it seems like porting a project originally started on KiCAD 8.0x broke KiBuzzard.

If I apply changes to labels or create new ones changes do not appear.

Any ideas/plans to give the add-on a little update?

Cheers

🍻

gregdavill, 2025-02-25:
I've not had a chance to dig into any specific changes needed for KiCad 9. 

It's on my list to update the plugin when I start using KiCad 9

dzalf, 2025-02-25:
Thanks, Greg.

No pressure, though.

Thanks for helping us all with this fantastic tool

🍻

pm-mic, 2025-02-27:
> I'm running this on v9.0.0. Adding `9.` to the version checks on L64 & L124 in `plugin.py` and it just worked. I've edited and added labels. Granted my testing was not very thorough - but it seems to work fine.

+1 Just tried this and it works. I'll test and see if I run in any problems.

Lambosaurus, 2025-02-27:
I see this has already been done at 5812f20314cbccec05ed7b099c1a2585e1e26141 - so we might just be waiting for a release.

gregdavill, 2025-03-01:
1.6.1 was released. A package to install via the PCM is in the release: https://github.com/gregdavill/KiBuzzard/releases/tag/1.6.1

It should be available directly in the PCM in the coming days. (Merged upstream: https://gitlab.com/kicad/addons/metadata/-/commit/205562d9d347971cfd21cb41aa943b5b5f165866 )