gregdavill/KiBuzzard #133: Not working in KiCad 9

KPkaustubhKP, 2025-02-27:
It is not working in KiCad 9, I am able to open the plugin and write my text but the moment i click on "ok" there is no silkscreen to be placed.

gregdavill, 2025-02-28:
In some limited initial testing it's been working for me, but only on the 2nd label. The first doesn't appear. 

When this happens are you able to "Paste" the label onto the board? It should end up in the clipboard.

fpp3, 2025-02-28:
Same here. I also can't edit text generated with the KiCad 8 version of the plugin

KPkaustubhKP, 2025-02-28:
No even 2nd label doesn't appear for me.
no it does not go into my clipboard as whenever i "Paste" it does nothing (pastes some random text i had copied)

gregdavill, 2025-02-28:
What platform are you on, and which version of the plugin are you using?

In the plugin folder there should be a `kibuzzard.log` which might have some useful errors or information when it's trying to insert the footprint into the design.

jjcarrol, 2025-03-01:
Also not working for me on KiCad 9.0. Nothing goes into the clipboard to paste.

 I'm running OSX 15.3.1. KIBuzzard is ver 1.6.0.

This is the only thing that's relevant in the kibuzzard.log file:

```
03-01 12:29:26 com_github_gregdavill_KiBuzzard.plugin 57:Running KiBuzzard
03-01 12:29:26 com_github_gregdavill_KiBuzzard.plugin 111:Version check failed "9.0.0" not in version list
03-01 12:29:26 com_github_gregdavill_KiBuzzard.plugin 152:Version check failed "9.0.0" not in version list
```

gregdavill, 2025-03-01:
Version 1.6.0 was released prior to KiCad 9. 
 - To ensure compatibility we check the version before inserting the label.
Please update to version 1.6.1. Which adds version 9 to this version check.

A package to install via the PCM is in the release: https://github.com/gregdavill/KiBuzzard/releases/tag/1.6.1

It should be available as a release on the PCM in the coming days. (Merged upstream: https://gitlab.com/kicad/addons/metadata/-/commit/205562d9d347971cfd21cb41aa943b5b5f165866 )

jjcarrol, 2025-03-01:
@gregdavill Just installed 1.6.1 via the PCM file and everything is working again on my 9.0 install. Nice job!

KPkaustubhKP, 2025-03-02:
Version 1.6.1 works

closed 2025-03-02 by KPkaustubhKP